SRJC-Computer-Science-Club/micromouse #15: Add a Mouse class

whothat, 2016-02-28:
a class that represents the mouse.
It will at a minimum consist of a position, direction, and functions to move around.

joshuasrjc, 2016-03-03:
I developed a class, and used the name MouseBot. It consists of position, direction, and functions to move around. I'm planning on adding functions to get input from the sensors, once we have a working "virtual maze."

RosyMapleMoth, 2016-03-03:
Hay Joshua do you have it in a state where you can put up a pull request I've been working with a sudo mosueBot class to get the core mapping functionality down but it would be nice if I could just switch over to building it around your real mouseBot class.

RosyMapleMoth, 2016-03-03:
Also MouseBot will need to have two functions that return the direction to the left and right of mouseBot.

RosyMapleMoth, 2016-03-04:
MouseBot has been merged in to dev

closed 2016-03-04 by RosyMapleMoth