SRJC-Computer-Science-Club/micromouse #23: Add core mapping functionality

whothat, 2016-03-02:
starting with the assumption that hardware will have a function called moveInDirection( Vector::Dir vec ) which allows you to move a given number of nodes in any direction n,s,w,e,nw, etc. 

write a function that will .
-move forward 1 node using moveInDIrection
-check the number of ways the mouse could go from this cell (eg there is a wall infront but no walls to the side, then you have 2 possible ways to go) - assume you have working functions that can check for walls
-if more than one possible way to go
    -add this node to a list (first in last out)
    -pick one direction to go
-if no possible ways to go
   -pathfind back to a node on the list
-repeat

RosyMapleMoth, 2016-03-05:
Will we be able to get readings from the IR transmitters for the NW and NE nodes from our facing as well as the W, N, E  nodes?

whothat, 2016-03-05:
If you are in the center of a cell it is safe to assume you can sense N, W, E there is no need to check NW or NE.

I think it might make the most sense to let the hardware controller deal with all sensor data. So for mapping you would just call a function to see if there is a wall. What do you guys think?

RosyMapleMoth, 2016-03-05:
I agree that's what i've been doing. as for the NW NE question I figured out a better solution to the problem having those would fix. PR : should have this done by tomorrow night maybe midday sunday.

whothat, 2016-03-05:
can you explain what how you would use the NW NE?

closed 2016-03-29 by whothat